emelianov/modbus-esp8266 #294: ESP 32 - MODBUS TCP SLAVE - Error

drlukbgd, 2023-06-15:
Hi, 

I made code for ESP32 using example code for MODBUS TCP SLAVE. Actual function of ESP32 is slave device with SPI interface to Wiz5500 Ethernet. As MODBUS MASTER TCP I used several different simulators. 

1. https://en.radzio.dxp.pl/modbus-master-simulator/
2. https://sourceforge.net/projects/qmodmaster/
3. https://www.winmodbus.com/

in all simulators after certain time appear error **Modbus message time out**  : 
[image]
[image]
 

Below is code : 

[ESP32_MODBUS_TCP_SLAVE.txt](https://github.com/emelianov/modbus-esp8266/files/11752015/ESP32_MODBUS_TCP_SLAVE.txt)

I will be very glad if someone can send me guidelines in order to solve this issue.   

Thanks a lot.

Merdock1, 2023-06-20:
Hi, 2 things that caught my attention:
1 - You do not have read and write tasks inside the modbus loop.
2 - Try not to use delay() inside the loop.
Within the loop you must associate reading and writing tasks of the type of data you use, here are examples of what I am talking about.

[https://github.com/emelianov/modbus-esp8266/blob/master/examples/TCP-Ethernet/client/client.ino](https://github.com/emelianov/modbus-esp8266/blob/master/examples/TCP-Ethernet/client/client.ino)

drlukbgd, 2023-06-21:
Hi, 

thanks for comments. After removing delay() in loop slave behavior is perfect without error message and lost packages. Great library !!!

closed 2023-10-14 by emelianov