emelianov/modbus-esp8266 #296: Modbus Client TCP

Barone1968, 2023-07-04:
Thank you for the help you provide to all of us, I have a problem that certainly others have already solved, I use the TCP example client code that connects correctly with the Server, the Client on monitor always returns me 0 even if the value of the registers varies on the Server, my need is to read 10 registers from server t and have them individually available on the client.... example first register read = Var 1, second read on Var 2 etc. etc. can you help me thanks.

This is the code that USE

/*
Modbus-Arduino Example - Master Modbus IP Client (ESP8266/ESP32)
Read Holding Register from Server device
(c)2018 Alexander Emelianov ([a.m.emelianov@gmail.com](mailto:a.m.emelianov@gmail.com))
https://github.com/emelianov/modbus-esp8266
*/

#ifdef ESP8266
#include <ESP8266WiFi.h>
#else
#include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

const int REG = 528; // Modbus Hreg Offset
IPAddress remote(192, 168, 199, 233); // Address of Modbus Slave device
const int LOOP_COUNT = 10;

ModbusIP mb; //ModbusIP object

void setup() {
Serial.begin(115200);

WiFi.begin("xxx", "111");

while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}

Serial.println("");
Serial.println("WiFi connected");
Serial.println("IP address: ");
Serial.println(WiFi.localIP());

mb.client();
}

uint16_t res = 0;
uint8_t show = LOOP_COUNT;

void loop() {
if (mb.isConnected(remote)) { // Check if connection to Modbus Slave is established
mb.readHreg(remote, REG, &res, 10); // Initiate Read Hregs from Modbus Server
} else {
mb.connect(remote); // Try to connect if no connection
}
mb.task(); // Common local Modbus task
delay(100); // Pulling interval
if (!show--) { // Display Slave register value one time per second (with default settings)
Serial.println(res);
show = LOOP_COUNT;
}
}

ChangC2, 2023-10-27:
I also have the same issue while using the Modbus TCP client.
Read all values 0.
Did you find the reason and solution?

emelianov, 2023-10-29:
Try to use more complex example that will show error code.
https://github.com/emelianov/modbus-esp8266/blob/master/examples/Callback/Transactional/Transactional.ino

ChangC2, 2023-10-30:
Thank you for your comment, Alexander Emelianov.
I used the sample you provided in this url, 
https://github.com/emelianov/modbus-esp8266/blob/master/examples/TCP-ESP/client/client.ino.

#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

const int REG = 528;               // Modbus Hreg Offset
IPAddress remote(192, 168, 30, 13);  // Address of Modbus Slave device
const int LOOP_COUNT = 10;

ModbusIP mb;  //ModbusIP object

void setup() {
  Serial.begin(115200);
 
  WiFi.begin("SSID", "PASSWORD");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.client();
}

uint16_t res = 0;
uint8_t show = LOOP_COUNT;

void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    mb.readHreg(remote, REG, &res);  // Initiate Read Coil from Modbus Slave
  } else {
    mb.connect(remote);           // Try to connect if no connection
  }
  mb.task();                      // Common local Modbus task
  delay(100);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    Serial.println(res);
    show = LOOP_COUNT;
  }
}

I tested with the EasyModbusTCP Server Simulator.
When I test the sample, success to connect to server.
but when  I read the hreg, it always get the Zero.
Could you provide the sample code that can read the href correctly?

ChangC2, 2023-10-30:
Hello, Alexander Emelianov. How are you?Thank you for your help.I used the sample you provided in this url,https://github.com/emelianov/modbus-esp8266/blob/master/examples/TCP-ESP/client/client.ino.#ifdef ESP8266#include <ESP8266WiFi.h>#else#include <WiFi.h>#endif#include <ModbusIP_ESP8266.h>const int REG = 528; // Modbus Hreg OffsetIPAddress remote(192, 168, 30, 13); // Address of Modbus Slave deviceconst int LOOP_COUNT = 10;ModbusIP mb; //ModbusIP objectvoid setup() {Serial.begin(115200);WiFi.begin("SSID", "PASSWORD");while (WiFi.status() != WL_CONNECTED) {delay(500);Serial.print(".");}Serial.println("");Serial.println("WiFi connected");Serial.println("IP address: ");Serial.println(WiFi.localIP());mb.client();}uint16_t res = 0;uint8_t show = LOOP_COUNT;void loop() {if (mb.isConnected(remote)) { // Check if connection to Modbus Slave is establishedmb.readHreg(remote, REG, &res); // Initiate Read Coil from Modbus Slave} else {mb.connect(remote); // Try to connect if no connection}mb.task(); // Common local Modbus taskdelay(100); // Pulling intervalif (!show--) { // Display Slave register value one time per second (with default settings)Serial.println(res);show = LOOP_COUNT;}}I tested with the EasyModbusTCP Server Simulator.When I test the sample, success to connect to server.but when I read the hreg, it always get the Zero.Could you provide the sample code that can read the href correctly? Thank youChang Cong Cong Sent from Mail for Windows From: Alexander EmelianovSent: Sunday, October 29, 2023 11:09 AMTo: emelianov/modbus-esp8266Cc: ChangC2; CommentSubject: Re: [emelianov/modbus-esp8266] Modbus Client TCP (Issue #296) Try to use more complex example that will show error code.https://github.com/emelianov/modbus-esp8266/blob/master/examples/Callback/Transactional/Transactional.ino

wupeng1987824, 2023-12-29:
> Thank you for your comment, Alexander Emelianov. I used the sample you provided in this url, https://github.com/emelianov/modbus-esp8266/blob/master/examples/TCP-ESP/client/client.ino.
>
> #ifdef [...]

because the slaveId is fixed to 255. I have the troubles too. how to config the slaveId?

wupeng1987824, 2023-12-29:
> Thank you for your comment, Alexander Emelianov. I used the sample you provided in this url, https://github.com/emelianov/modbus-esp8266/blob/master/examples/TCP-ESP/client/client.ino.
>
> #ifdef [...]

change the read function to below:  [unit]  instead of your master slavId. default 255 
mb.readHreg(remote, REG, &res,1,nullptr,1);

closed 2024-05-11 by emelianov