emelianov/modbus-esp8266 #314: Issues with writing large data blocks (60+ registers) when using ESP8266 as master

SamZaitcev, 2023-10-17:
Hello,

I've been using your **modbus-esp8266** library with an ESP8266 controller and encountered an issue when trying to write data blocks of 60 or more registers at once. When writing smaller blocks (e.g., 10 registers), everything works fine. I'm building my project using PlatformIO.

**Steps to reproduce:**
1. Configure ESP8266 as a master.
2. Attempt to write a data block of 60 or more registers.

**Expected behavior:**
Data gets written successfully and the callback is executed.

**Current behavior:**
The callback isn't executed, and mb.slave() gets stuck at true.

**Additional information:**
All other controller functions, not related to Modbus, continue to work as expected.

Any assistance or advice on this issue would be greatly appreciated!

emelianov, 2023-10-19:
As far as I understand you are using ModbusRTU. Can you provide extra info as
- line speed used
- SoftwareSerial library version used

Also quick suggestion is to increase SoftwareSerial buffers size.

SamZaitcev, 2023-10-19:
Yes, I am using ModbusRTU. The settings are 9600, 8, 'N', 2.
The version of SoftwareSerial I'm using is 8.0.1.

SamZaitcev, 2023-10-19:
I've set the buffer to 200 bytes using:

```
S.begin(modbus_configuration->getBoundRate(),
        getSerialConfig(modbus_configuration->getStopBits(),
                        modbus_configuration->getParity(),
                        modbus_configuration->getDataBits()),
        -1,
        -1,
        false,
        200);

```
However, the behavior remains unchanged. Any further suggestions?
Thanks.

emelianov, 2023-10-29:
Anyway, try to specify SoftwareSerial buffer size:
https://github.com/emelianov/modbus-esp8266/issues/177#issuecomment-1011077322

emelianov, 2023-10-29:
Also you may play with SoftwareSerial version. The issue is related to SoftwareSerial as it's not so stable on data send\receive as HardwareSerial.
(Tests of read Hegs runs okay for up to 126 regs with no SoftwareSerial)

closed 2024-04-30 by emelianov