emelianov/modbus-esp8266 #323: #define MODBUSRTU_REDE leads to compiler errors

fredlcore, 2023-12-08:
It's not an actual error, but current default settings in Arduino IDE lead to this error when compiling with MODBUSRTU_REDE enabled:
```
/Users/frederik/Documents/Arduino/libraries/modbus-esp8266/src/ModbusRTU.cpp: In member function 'bool ModbusRTUTemplate::rawSend(uint8_t, uint8_t*, uint8_t)':
/Users/frederik/Documents/Arduino/libraries/modbus-esp8266/src/ModbusRTU.cpp:138:6: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
      if (_txPin >= 0)
      ^~
/Users/frederik/Documents/Arduino/libraries/modbus-esp8266/src/ModbusRTU.cpp:140:3: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
   if (_rxPin >= 0)
   ^~
/Users/frederik/Documents/Arduino/libraries/modbus-esp8266/src/ModbusRTU.cpp:167:6: error: this 'if' clause does not guard... [-Werror=misleading-indentation]
      if (_txPin >= 0)
      ^~
/Users/frederik/Documents/Arduino/libraries/modbus-esp8266/src/ModbusRTU.cpp:169:3: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
   if (_rxPin >= 0)
   ^~
cc1plus: some warnings being treated as errors
/Users/frederik/Documents/Arduino/libraries/modbus-esp8266/src/Modbus.cpp: In member function 'void Modbus::slavePDU(uint8_t*)':
/Users/frederik/Documents/Arduino/libraries/modbus-esp8266/src/Modbus.cpp:318:25: warning: comparison is always false due to limited range of data type [-Wtype-limits]
             if (bufSize > MODBUS_MAX_FRAME) {  // Frame to return too large

Using library modbus-esp8266 at version 4.1.0 in folder: /Users/frederik/Documents/Arduino/libraries/modbus-esp8266 
exit status 1

Compilation error: exit status 1
```

It's really just a matter of indention. By moving the commands into the same line as the if-statement, the error disappears.

rob040, 2023-12-15:
This is caused by that this part of code has a mix of spaces and TABs as indent, where the TAB size setting is different from what the code maintainer has uses.
Convert TAB to spaces and proper formatting the code will solve warnings. 
(someone has instructed gcc to treat warnings as errors... its a certain way to get those pesky warnings get fixed before anyone can continue...)

fredlcore, 2023-12-15:
I know, and I fixed it myself already. I just wanted to highlight this to be fixed in this repo as well so that others won't run into this problem on a standard Arduino IDE installation.

closed 2024-05-11 by emelianov