emelianov/modbus-esp8266 #328: strange code received by ESP32

marcovilla118, 2024-01-05:
Hi,

I tried to run esp32 as modbus slave.

MASTER send:

01 03 00 00 00 0A C5 CD

SLAVE (esp32 with basic RTU sketch) send back:

7F 7E FF FF EB 75 19

This is really strange and I tried to modify the slave address and other parameter but with no success. Do you have any idea about this code, thanks!!!

marcovilla118, 2024-01-05:
here the part of the code:

#define RX 36
#define TX 37
#define SLAVE_ID 1
//#define MODBUSRTU_REDE
ModbusRTU mb;

void setup() {
  Serial.begin(9600,SERIAL_8N1,RX,TX);

after some tests I found that changing the PIN number (rx, tx) doesn't change the output. So it looks like the hardware pin I have assigned are not used by the software.

emelianov, 2024-04-26:
https://github.com/emelianov/modbus-esp8266/blob/master/documentation/README.md#modbusrtu

closed 2024-05-27 by emelianov