emelianov/modbus-esp8266 #348: How to write single register

legaxus, 2024-05-03:
Appreciated if you could show and guide me how to write single register, thanks.

Oaluiser, 2024-05-03:
If you are a server: Hreg(register, value)
If you are a client: writeHreg(address, register, value)

legaxus, 2024-05-03:
> If you are a server: Hreg(register, value) If you are a client: writeHreg(address, register, value)

Thanks for the guidance, just to clarify, server meaning is Master, client meaning is Slave?
For Slave/Client, Address = Slave ID?

Thanks in advanced.

emelianov, 2024-05-04:
> Thanks for the guidance, just to clarify, server meaning is Master, client meaning is Slave? For Slave/Client, Address = Slave ID?

Client is Master, Server is Slave
Address is SlaveID

legaxus, 2024-05-06:
Thanks for the clarification. I've tried and got this error, value = 1 NO error, but if value = 0, compilation error. Attached with picture. Thanks.
[image]

legaxus, 2024-05-07:
I've tried with mb.writeHreg(slaveid, 5, 1, NULL) it managed to transmit value of 1 to slave, then I tried with mb.writeHreg(slaveid, 5, 0, NULL), it just hang there without transmitting the 0 to slave. Compilation is success. Is there anything I had missed out?
Thanks.

legaxus, 2024-05-07:
> > Thanks for the guidance, just to clarify, server meaning is Master, client meaning is Slave? For Slave/Client, Address = Slave ID?
> 
> Client is Master, Server is Slave Address is SlaveID

I've tried with mb.writeHreg(slaveid, 5, 1, NULL) it managed to transmit value of 1 to slave, then I tried with mb.writeHreg(slaveid, 5, 0, NULL), it just hang there without transmitting the 0 to slave. Compilation is success. Is there anything I had missed out?
Thanks.

MarcinanProfesjonalista, 2024-09-23:
`#include <ModbusRTU.h>
#define SERIAL2_RX_PIN 16
#define SERIAL2_TX_PIN 17
float targetPosition = 0.00;
unsigned long previousMillis_rs = 0; // Przechowuje czas poprzedniej aktualizacji
const long interval = 1000; // Interwał w milisekundach (1 sekunda)
 `

`// Funkcje pomocnicze
void floatToRegisters(float value, uint16_t* regs) {
  union {
    float f;
    uint16_t regs[2];
  } data;
  data.f = value;
  regs[0] = data.regs[0];
  regs[1] = data.regs[1];
}`

` // Pakowanie `targetPosition` do rejestrów Modbus
  floatToRegisters(targetPosition, tempRegs);
  mb.Hreg(REG_TARGET_POSITION, tempRegs[0]);
  mb.Hreg(REG_TARGET_POSITION + 1, tempRegs[1]);`

`void init_modbus_rtu() {

  Serial2.begin(9600, SERIAL_8N1, SERIAL2_RX_PIN, SERIAL2_TX_PIN);
  mb.begin(&Serial2);
  mb.slave(id);

  // Dodawanie rejestrów
  mb.addHreg(REG_ID, id);  // Rejestr 0: ID
  mb.addCoil(REG_PENDING_FLAG, 0);
  mb.addHreg(REG_TARGET_POSITION, 12, 1);       // Rejestr 1-2: Cel podróży (float)
  mb.addHreg(REG_TARGET_POSITION+1, 34, 1);       // Rejestr 1-2: Cel podróży (float)
}`

`      unsigned long currentMillis = millis();
    
    // Sprawdzanie, czy minęła jedna sekunda
    if (currentMillis - previousMillis_rs >= interval) {
        previousMillis_rs = currentMillis; // Aktualizacja czasu

        targetPosition += 1.1; // Zwiększ wartość o 0.1

          // Pakowanie `targetPosition` do rejestrów Modbus
  floatToRegisters(targetPosition, tempRegs);
  mb.writeHreg(1,REG_TARGET_POSITION, tempRegs[0]);
  mb.writeHreg(1,REG_TARGET_POSITION + 1, tempRegs[1]);
        //Serial.println(targetPosition); // Wypisanie wartości na Serial Monitor
    }`

I'm on slave. My slave is holding own float value targetPosition. I cant upgade it on shared modbus register.

MarcinanProfesjonalista, 2024-09-23:
How the slave update its own modbus registers?

emelianov, 2024-11-24:
Handling of long/float over Modbus is described in #158