emelianov/modbus-esp8266 #349: Working on ESP32-S3, but the priority must be set with 0,  otherwise the CPU will falt to reset。

kylongmu, 2024-05-07:
`  
xTaskCreatePinnedToCore(
    taskModbus,   
    "taskModbus",  
    1 << 14,        
    NULL,         
    0,             //Must be 0
    NULL,         
    0);  //ARDUINO_RUNNING_CORE is 1

}

void taskModbus(void* parameter) {
  Serial1.begin(9600, SERIAL_8N2, RX_PIN, TX_PIN);
  mb.begin(&Serial1);
  mb.slave(Hreg_file[0]);
  mb.addHreg(0, 0, 16);
  mb.addHreg(Hreg_address, 0, Hreg_size);
  //Serial.printf("Running task taskModbus%d\n", xPortGetCoreID());
  while (1) {
    mb.task();
    yield();
  }
  Serial.println("Ending task taskModbus");
  vTaskDelete(NULL);
}

`
When the priority set to others, CPU reboot with this output:

E (10135) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10135) task_wdt:  - IDLE (CPU 0)
E (10135) task_wdt: Tasks currently running:
E (10135) task_wdt: CPU 0: taskModbus
E (10135) task_wdt: CPU 1: IDLE
E (10135) task_wdt: Aborting.

abort() was called at PC 0x420122b4 on core 0

MeisterQ, 2024-05-16:
Is there any work in progress here? I came to this issue aswell.

emelianov, 2024-05-17:
Waiting of -s3 board to arrive

M4E5TR0-MUN1R, 2024-08-16:
Hello, 

* The way to solve this issue is to replace the `yield();` with `vTaskDelay(pdMS_TO_TICKS(100));` 
* Explanation: `yield();` will only allow higher priority tasks to run, but the watchdog runs in the idle task (priority 0) so it won't get a chance to run if the taskModbus has a higher priority.
* Long Story Short: If you want taskModbus to run at a higher priority than 0. Dont use `yield();` function.