emelianov/modbus-esp8266 #350: I would like to use a large number of callback calls (more than 300)

kellers76, 2024-05-10:
The library is really great and I get on very well with it.
I have a lot of registers in my project and therefore also a lot of callback calls.
At the moment it seems that a limit has been reached at 300 callback calls (onGetCoil and onGetHreg combined). I get an error in the Arduino IDE but without it showing exactly where the problem is. Can you maybe help me to extend the maximum calls? At the moment I need 333, but a limit of 500 would be better.

I would appreciate your feedback

Translated with DeepL.com (free version)

emelianov, 2024-05-11:
Can you provide code causing problem and information on Arduino board you are using?
ESP32 compiles and runs okay (just test code to create 400 registers and callbacks):
```cpp
  for (uint16_t i = 0; i < 400; i++) {
    mb.addCoil(i);       // Add Coil. The same as mb.addCoil(COIL_BASE, false, LEN)
    mb.onSetCoil(i, cbLed); // Add callback on Coil LED_COIL value set
    if (i % 10 == 0) {
      Serial.print(i);
      Serial.print(": ");
      Serial.println(esp_get_free_heap_size());
    }
  }
```

kellers76, 2024-05-11:
Hello,

thanks for your quick feedback...

I am using an ESP32-Wroom-32E with the Arduino IDE and a memory table for 3MB code, 3MB OTA and 6MB SPIFFS.

I am not sure at the moment if I have done everything right or am doing it right. 
I'm not using mb.onSetCoil for the callbacks like you are but individual calls to Modbus addresses using mb.onGetCoil and mb.onGetHreg. This allows me to catch every single change and process it in different ways.

Here is an excerpt from the program code as an example:
              mb.onGetHreg(901, Modbusm_pElemVal_Einstell_Temperatur_Mo1, 1);
              mb.onGetHreg(912, Modbusm_pElemVal_Einstell_Temperatur_Mo2, 1);
              mb.onGetHreg(920, Modbusm_pElemVal_Einstell_Temperatur_Mo3, 1);
              mb.onGetHreg(930, Modbusm_pElemVal_Einstell_Temperatur_Di1, 1);
              mb.onGetHreg(941, Modbusm_pElemVal_Einstell_Temperatur_Di2, 1);
              mb.onGetHreg(952, Modbusm_pElemVal_Einstell_Temperatur_Di3, 1);
              mb.onGetHreg(963, Modbusm_pElemVal_Einstell_Temperatur_Mi1, 1);
              mb.onGetHreg(974, Modbusm_pElemVal_Einstell_Temperatur_Mi2, 1);
              mb.onGetHreg(985, Modbusm_pElemVal_Einstell_Temperatur_Mi3, 1);
              mb.onGetHreg(996, Modbusm_pElemVal_Einstell_Temperatur_Do1, 1);
              .
              .
              .

I am not a professional, so please excuse me if I am not using the best syntax or have expressed myself incorrectly

Translated with DeepL.com (free version)

emelianov, 2024-05-11:
Unable to reproduce error even generating code of 400 separate onGet calls.
Arduino-ESP32 version tested is 2.14, Arduino IDE 1.8.19

kellers76, 2024-10-18:
Hello

please excuse me that I am only now addressing the problem again.

I tried again to derive the problem and at the moment I'm not sure where the problem really is.

I haven't given you any important information yet.
The compiler message is:
... dangerous relocation: l32r: literal target out of range (try using text-section-literals): .literal._ZN2fs4FileD5Ev
...

This error message comes when the number of callbacks is around 300.
As you saw in my previous comment, I make individual calls to special functions because each CallBack is handled differently.

My CallBack calls are in the setup area and the routines that are then executed are in an extra .h file.
Maybe I'll lay an egg here because of my constellation, but I won't get anywhere without your professional help...

thanks!!

emelianov, 2024-11-24:
It may be related to specific code that facing resource limit of the board.