emelianov/modbus-esp8266 #352: Ignore Callbacks when Register is Updated by Self

seikosantana, 2024-05-25:
Hi, is there a way to update register value without triggering the onSetCoil / onSetHreg?
I am trying to integrate modbus into existing working program so that it can interface with modbus, but the problem is when I use onSetCoil/onSetHreg to handle incoming requests, and have my program to update the registers to represent the current state, it fires the callbacks too, causing recursive calls or long loops.

Do you have any tips or workarounds for this kind of situation? Thanks in advance

emelianov, 2024-05-26:
```c
mb.cbDisable();
mb.Hreg(REV, VALUE);
mb.cbEnable();
```

seikosantana, 2024-05-27:
Is it save to call those in a callback itself (inside an onSet)?

emelianov, 2024-05-27:
Sure.

seikosantana, 2024-05-27:
Thanks a lot for your response and effort on the library 😀

closed 2024-05-27 by seikosantana