emelianov/modbus-esp8266 #355: Question: is it possible to have multiple servers on the same ESP8266 / serial line ?

MichielfromNL, 2024-06-25:
HI,

I need some help with getting a multi-server configuration to work on ESP8266.
Setup: I emulate two DTSU666 3-phase power meters, for a power unit that is connected to my ESP8266/485MAX
The power unit subsequently reads on address 1 and address 2

When I run the sketch with 1 server instance (on address1) everything works fine. 
But when I add a second ModbusRTU server (address2) one the same line, I can't get it to work , and I can't figure out why not, I suspect that only one will read the serial line so data is "lost" for server 2 when server 1 decides that it is not adressed.
Hence the question: is it possible to have 2 servers running on the same ESP8266/MAX485, with different adresses?

Thanks!

emelianov, 2024-06-26:
For sure, multiple servers(slaves) are okay on single line.
If client(master) is also driven by the library code? That case could you share the code to review?

MichielfromNL, 2024-06-30:
HI,
I looked further into it, and it makes sense that it won't work for multiple slaves listening on the same serial port.
reason: modbusRTU.cpp lines 236 -245.  If there is data available on the serial port, the address is read. When not a valid address, the frame is discarded. And data which is read from a port can't be unread - and made available for another slave.
Solution: multiple max(3)485\s, connected both on the same AB RS485 lines, then it works like a charm.
Sorry for bothering you with this question.

emelianov, 2024-07-02:
Under 'serial line' I've assumed sharing physical line not device Serial port. The library object instances unable to share the same port. So your solution to use multiple ports|converters is right.
Also it's possible to configure the library object instance to answer on multiple SlaveID: https://github.com/emelianov/modbus-esp8266/blob/master/examples/Bridge/MultipleServerID/MultipleServerID.ino

closed 2024-10-27 by emelianov