emelianov/modbus-esp8266 #356: ModbusRTU result: E4, Mem: XXXXX RTU Timeout

Eduardo-bat, 2024-06-26:
Hi!

I'm testing the following code as a link between https://sourceforge.net/projects/qmodmaster/ and https://pypi.org/project/pyModSlave/. As hardware, I'm using a NodeMCU 1.0 and a USB/TTL converter.

```
#include <ESP8266WiFi.h>
#include <ModbusTCP.h>
#include <ModbusRTU.h>

ModbusRTU rtu;
ModbusTCP tcp;

IPAddress srcIp;

uint16_t transRunning = 0;
uint8_t slaveRunning = 0;

uint16_t ModbusBaudRate = 19200;

const char* ssid     = "<SSID>";
const char* password = "<PASSWORD>";

Modbus::ResultCode cbTcpRaw(uint8_t* data, uint8_t len, void* custom) {
  auto src = (Modbus::frame_arg_t*) custom;

  srcIp = src->ipaddr;
  slaveRunning = src->slaveId;
  transRunning = src->transactionId;

  rtu.rawRequest(slaveRunning, data, len);

  return Modbus::EX_SUCCESS;
}

Modbus::ResultCode cbRtuRaw(uint8_t* data, uint8_t len, void* custom) {
  tcp.setTransactionId(transRunning);
  tcp.rawResponce(srcIp, data, len, slaveRunning);

  transRunning = 0;
  slaveRunning = 0;
  srcIp = (0,0,0,0);
  return Modbus::EX_PASSTHROUGH;
}

void setup() {
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
    delay(1000);

  tcp.server();
  tcp.onRaw(cbTcpRaw);

  Serial.begin(ModbusBaudRate, SERIAL_8E1);
  rtu.begin(&Serial);
  rtu.master();
  rtu.onRaw(cbRtuRaw);
}

void loop() {
  rtu.task();
  tcp.task();
  yield();
}
```

When I make any request from the TCP client, I receive an RTU Timeout.

I've already tried using hard and soft Serial, baudrates of 9600 and 19200 and powering the board with a regulated supply @5 V-1 A.

Both programs work normally when I communicate them directly with the ESP, either reading from the RTU slave or receiving reads from the TCP client.

When I use ESP as a bridge, the RTU slave receives the request normally and responds correctly in < 5 ms, but the timeout occurs.

Please, any hint on what to test or what else to investigate? Thanks.

emelianov, 2024-06-27:
Probably this will help  #298.
Otherwise I'll back later

Eduardo-bat, 2024-06-29:
Hi @emelianov. Thank you for the direction. Haven't had time to thoroughly test it yet, but already gotten some promising results. When I get to proceed, I will provide more details.

Eduardo-bat, 2024-07-04:
Unfortunately, the problem persists when RTU is connected over a Serial.

My current code is

```
/*
  ModbusRTU ESP8266/ESP32
  ModbusTCP to ModbusRTU bridge with on-device ModbusRTU simulator
*/
#ifdef ESP8266
#include <ESP8266WiFi.h>
#else
#include <WiFi.h>
#endif

#include <ModbusTCP.h>
#include <ModbusRTU.h>

ModbusRTU rtu;
ModbusTCP tcp;

IPAddress srcIp;

uint16_t transRunning = 0;
uint8_t slaveRunning = 0;

bool cbTcpTrans(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  if (event != Modbus::EX_SUCCESS) {
    //Serial.printf("Modbus result: %02X, Mem: %d\n", event, ESP.getFreeHeap());
  } if (event == Modbus::EX_TIMEOUT) {
    tcp.disconnect(tcp.eventSource());
  }
  return true;
}

bool cbRtuTrans(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  if (event != Modbus::EX_SUCCESS) {
    //Serial.printf("Modbus result: %02X, Mem: %d\n", event, ESP.getFreeHeap());
  }
  return true;
}

Modbus::ResultCode cbTcpRaw(uint8_t* data, uint8_t len, void* custom) {
  auto src = (Modbus::frame_arg_t*) custom;
  
  //Serial.print("TCP IP in - ");
  //Serial.print(IPAddress(src->ipaddr));
  //Serial.printf(" Fn: %02X, len: %d \n\r", data[0], len);

  if (transRunning) {
    tcp.setTransactionId(src->transactionId);
    tcp.errorResponce(IPAddress(src->ipaddr), (Modbus::FunctionCode)data[0], Modbus::EX_SLAVE_DEVICE_BUSY);
    return Modbus::EX_SLAVE_DEVICE_BUSY;
  }

  rtu.rawRequest(src->unitId, data, len, cbRtuTrans);
  
  if (!src->unitId) {
    tcp.setTransactionId(src->transactionId);
    tcp.errorResponce(IPAddress(src->ipaddr), (Modbus::FunctionCode)data[0], Modbus::EX_ACKNOWLEDGE);

    transRunning = 0;
    slaveRunning = 0;
    return Modbus::EX_ACKNOWLEDGE;
  }
  
  srcIp = IPAddress(src->ipaddr);
  
  slaveRunning = src->unitId;
  
  transRunning = src->transactionId;
  
  return Modbus::EX_SUCCESS;  
  
}

Modbus::ResultCode cbRtuRaw(uint8_t* data, uint8_t len, void* custom) {
  auto src = (Modbus::frame_arg_t*)custom;
  tcp.setTransactionId(transRunning);
  uint16_t succeed = tcp.rawResponce(srcIp, data, len, slaveRunning);
  if (!succeed) {
    //Serial.print("fail");
  }
  //Serial.printf("RTU Slave: %d, Fn: %02X, len: %d, ", src->slaveId, data[0], len);
  //Serial.print("Response TCP IP: ");
  //Serial.println(srcIp);

  transRunning = 0;
  slaveRunning = 0;
  return Modbus::EX_PASSTHROUGH;
}

void setup() {
  Serial.begin(19200, SERIAL_8E1);
  WiFi.begin("<SSID>", "<PASSWORD>");
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    //Serial.print(".");
  }
  //Serial.println("");
  //Serial.println("IP address: ");
  //Serial.println(WiFi.localIP());

  tcp.server();
  tcp.onRaw(cbTcpRaw);

  rtu.begin(&Serial);
  rtu.master();
  rtu.onRaw(cbRtuRaw);
}

void loop() {
  rtu.task();
  tcp.task();
  yield();
}
```

My RTU slave, in pyModSlave, is configured as seen in the following image:
[image]

My TCP client, in qModMaster, is set up like this:
[image]

My RTU log is:

```
[RTU]>Rx > 09:56:29:369 - 01 03 00 01 00 01 D5 CA 
[RTU]>Tx > 09:56:29:369 - 01 03 02 49 39 4F C6 
```

My TCP log is:

```
Sys > 09:56:20:520 - Connecting to IP : 192.168.003.165:502 OK
[TCP]>Tx > 09:56:29:338 - 00  01  00  00  00  06  01  03  00  01  00  01  
Sys > 09:56:29:843 - Error : Timeout
[TCP]>Tx > 09:56:35:848 - 00  02  00  00  00  06  01  03  00  01  00  01  
[TCP]>Rx > 09:56:35:878 - 00  02  00  00  00  03  FF  83  06  
Sys > 09:56:35:878 - Error : Slave device or server is busy
[TCP]>Tx > 09:56:43:108 - 00  03  00  00  00  06  01  03  00  01  00  01  
[TCP]>Rx > 09:56:43:124 - 00  03  00  00  00  03  FF  83  06  
Sys > 09:56:43:124 - Error : Slave device or server is busy
```

Eduardo-bat, 2024-07-07:
By the following code, it seems like the problem happens before the library calls:

```
#include <ESP8266WiFi.h>

#include <ModbusTCP.h>
#include <ModbusRTU.h>

#include <SoftwareSerial.h>
#include <StreamBuf.h>

#define BSIZE 1024

const unsigned long baudrate = 19200;
const unsigned long chtimeus = 16 * 1000000 / baudrate;

ModbusRTU rtu;
ModbusTCP tcp;

IPAddress srcIp;

SoftwareSerial debug_serial(D1, D2);

uint8_t serialw_rtur[BSIZE];
uint8_t serialr_rtuw[BSIZE];

StreamBuf serialw_rtur_sb(serialw_rtur, BSIZE);
StreamBuf serialr_rtuw_sb(serialr_rtuw, BSIZE);

DuplexBuf modbus_db(&serialw_rtur_sb, &serialr_rtuw_sb);
DuplexBuf serial_db(&serialr_rtuw_sb, &serialw_rtur_sb);

uint8_t com_buf[BSIZE];
uint8_t log_buf[BSIZE];

StreamBuf com_sbuf(com_buf, BSIZE);
StreamBuf log_sbuf(log_buf, BSIZE);

unsigned long last_us = 0;

uint16_t transRunning = 0;
uint8_t slaveRunning = 0;

bool cbRtuTrans(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  if (event != Modbus::EX_SUCCESS) {
    debug_serial.printf("Modbus result: %02X, Mem: %d\n", event, ESP.getFreeHeap());
    transRunning = 0;
  }
  return true;
}

Modbus::ResultCode cbTcpRaw(uint8_t* data, uint8_t len, void* custom) {
  auto src = (Modbus::frame_arg_t*) custom;
  
  debug_serial.print("TCP IP in - ");
  debug_serial.print(IPAddress(src->ipaddr));
  debug_serial.printf(" Fn: %02X, len: %d \n\r", data[0], len);

  if (transRunning) {
    tcp.setTransactionId(src->transactionId);
    tcp.errorResponce(IPAddress(src->ipaddr), (Modbus::FunctionCode)data[0], Modbus::EX_SLAVE_DEVICE_BUSY);
    return Modbus::EX_SLAVE_DEVICE_BUSY;
  }

  rtu.rawRequest(src->unitId, data, len, cbRtuTrans);
  
  if (!src->unitId) {
    tcp.setTransactionId(src->transactionId);
    tcp.errorResponce(IPAddress(src->ipaddr), (Modbus::FunctionCode)data[0], Modbus::EX_ACKNOWLEDGE);

[... 184 lines cut ...]
received response: : 1 20 11 E 23 76 A CA 20 18 8D 66 58 69 A9 FF 1F 1D 93 FF 
TCP IP in - 192.168.1.13 Fn: 03, len: 5 

Modbus result: E4, Mem: 41336
TCP IP in - 192.168.1.13 Fn: 03, len: 5 

sent request: : 1 3 0 0 0 A C5 CD 
received response: : 1 20 51 A BE 77 23 68 A E5 6B 4B 6D 71 84 5D 3A 7C E9 
TCP IP in - 192.168.1.13 Fn: 03, len: 5 

Modbus result: E4, Mem: 42008
TCP IP in - 192.168.1.13 Fn: 03, len: 5 

sent request: : 1 3 0 0 0 A C5 CD 
received response: : 1 20 11 74 5B F9 D7 AC 11 BD 97 
received response: : DB 7E A5 CB BD 54 2F B0 FF 
TCP IP in - 192.168.1.13 Fn: 03, len: 5 

Modbus result: E4, Mem: 41336
TCP IP in - 192.168.1.13 Fn: 03, len: 5 

sent request: : 1 3 0 0 0 A C5 CD 
received response: : 1 20 91 36 A D8 AF 5E A2 C2 6D 37 BB 2C 8 32 F5 7E D2 FE 
Modbus result: E4, Mem: 42008
```

Eduardo-bat, 2024-07-14:
An external monitor (a Uno with the Rx of a `SoftwareSerial` connected to the NodeMCU hardware Rx pin) shows that the correct data reaches the ESP:

rtu monitor:
```
[RTU]>Rx > 18:21:37:626 - 01 03 00 00 00 0A C5 CD 
[RTU]>Tx > 18:21:37:626 - 01 03 14 9D AC D8 0D 61 1A 29 7B 0E 7D 19 15 62 01 C2 F1 9E CF 41 C6 3C D4 
[RTU]>Rx > 18:21:47:674 - 01 03 00 00 00 0A C5 CD 
[RTU]>Tx > 18:21:47:675 - 01 03 14 A7 EE 7E F2 5D 43 66 D7 4B 73 3A 0A D2 B6 23 BB F0 E4 AB D0 CD 9E 
[RTU]>Rx > 18:21:57:590 - 01 03 00 00 00 0A C5 CD 
[RTU]>Tx > 18:21:57:590 - 01 03 14 D0 31 DC 14 E7 A2 84 49 C6 3A 1D 13 BD 7E DD DF E1 C7 02 89 F0 80 
[RTU]>Rx > 18:22:07:638 - 01 03 00 00 00 0A C5 CD 
[RTU]>Tx > 18:22:07:639 - 01 03 14 71 26 FE 5B 12 23 A7 F4 39 0A 55 12 F9 E4 63 27 90 0C 18 11 00 51 
[RTU]>Rx > 18:22:17:682 - 01 03 00 00 00 0A C5 CD 
[RTU]>Tx > 18:22:17:683 - 01 03 14 1C 9D A1 68 11 71 A4 96 1D BF D4 C3 86 B1 B2 42 68 C4 72 BD C6 A3 
[RTU]>Rx > 18:22:27:722 - 01 03 00 00 00 0A C5 CD 
[RTU]>Tx > 18:22:27:724 - 01 03 14 93 6B 28 A4 F2 DD 3D E1 C3 29 61 82 EF CC E3 37 99 AA 32 04 3A 56 
```

external monitor:
```
response: 1 3 14 9D AC D8 D 61 1A 29 7B E 7D 19 15 62 1 C2 F1 9E CF 41 C6 3C D4
response: 1 3 14 A7 EE 7E F2 5D 43 66 D7 4B 73 3A A D2 B6 23 BB F0 E4 AB D0 CD 9E
response: 1 3 14 D0 31 DC 14 E7 A2 84 49 C6 3A 1D 13 BD 7E DD DF E1 C7 2 89 F0 80
response: 1 3 14 71 26 FE 5B 12 23 A7 F4 39 A 55 12 F9 E4 63 27 90 C 18 11 0 51
response: 1 3 14 1C 9D A1 68 11 71 A4 96 1D BF D4 C3 86 B1 B2 42 68 C4 72 BD C6 A3
response: 1 3 14 93 6B 28 A4 F2 DD 3D E1 C3 29 61 82 EF CC E3 37 99 AA 32 4 3A 56
```

bridge monitor:
```
received a TCP request from 192.168.1.12-> Fn: 03, len: 5
sent request to RTU: 1 3 0 0 0 A C5 CD 
received response from RTU: 1 20 51 C6 C3 D 58 A3 69 39 7D 46 91 5 C2 7C F6 41 1C A2 
received a TCP request from 192.168.1.12-> Fn: 03, len: 5
modbusRTU transaction result: E4

received a TCP request from 192.168.1.12-> Fn: 03, len: 5
sent request to RTU: 1 3 0 0 0 A C5 CD 
received response from RTU: 1 20 D1 73 7E BE 35 99 D7 A9 E9 A 3A DB DA F0 DE D D7 9E 

received a TCP request from 192.168.1.12-> Fn: 03, len: 5
modbusRTU transaction result: E4

received a TCP request from 192.168.1.12-> Fn: 03, len: 5
sent request to RTU: 1 3 0 0 0 A C5 CD 
received response from RTU: 1 20 11 17 E2 14 9E 23 49 9C EA 13 6F 75 DF 78 9C 89 80 
received a TCP request from 192.168.1.12-> Fn: 03, len: 5
modbusRTU transaction result: E4

received a TCP request from 192.168.1.12-> Fn: 03, len: 5
sent request to RTU: 1 3 0 0 0 A C5 CD 
received response from RTU: 1 20 51 99 FE 2B 34 3A F4 4E A1 95 94 E4 AC 12 C6 18 44 51 
received a TCP request from 192.168.1.12-> Fn: 03, len: 5
modbusRTU transaction result: E4

received a TCP request from 192.168.1.12-> Fn: 03, len: 5
sent request to RTU: 1 3 0 0 0 A C5 CD 
received response from RTU: 1 20 11 D4 A1 2B 14 91 96 47 A3 C3 58 93 42 8B 72 6F DC FF 
received a TCP request from 192.168.1.12-> Fn: 03, len: 5
modbusRTU transaction result: E4

received a TCP request from 192.168.1.12-> Fn: 03, len: 5
sent request to RTU: 1 3 0 0 0 A C5 CD 
received response from RTU: 1 20 D1 B6 A1 A4 BE D7 A C3 4A 9 EF 6E 37 66 35 22 3A F5 
received a TCP request from 192.168.1.12-> Fn: 03, len: 5
modbusRTU transaction result: E4
```