emelianov/modbus-esp8266 #357: An error occurs in the header file when compiling.

kimwoo0, 2024-07-10:
What I did was install Arduino IDE 2.3.2 and installed the esp8266 by ESP8266 Communuty package.
Then I installed the library and loaded the example file and compiled it, but the following error occurred.

AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/lwip2/include/lwip/ip4_addr.h:63: warning: "IPADDR_NONE" redefined
modbus-esp8266\src/ModbusTCPTemplate.h:15: note: this is the location of the previous definition
   15 | #define IPADDR_NONE ((uint32_t)0xffffffffUL)

modbus-esp8266\src/ModbusTCPTemplate.h:17:1: error: 'Callback' does not name a type; did you mean 'TCallback'?

modbus-esp8266\src/ModbusTCPTemplate.h:20:23: error: 'IPAddress' was not declared in this scope; did you mean 'TAddress'?

I've encountered many similar types of errors.
What's wrong?

kimwoo0, 2024-07-11:
Most of the issues seem to have been resolved by changing the include order of the header file. However, the content below appears to be a different problem. Is there any solution?

In file included from c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusEthernet.h:12,
                 from C:\Users\kimwoo0.KOMECO\Documents\Arduino\sketch_jul10a\sketch_jul10a.ino:10:
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h:17:1: error: 'Callback' does not name a type; did you mean 'TCallback'?
   17 | Callback function Type
      | ^~~~~~~~
      | TCallback
In file included from c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusEthernet.h:12,
                 from C:\Users\kimwoo0.KOMECO\Documents\Arduino\sketch_jul10a\sketch_jul10a.ino:10:
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h:51:2: error: 'cbModbusConnect' does not name a type
   51 |  cbModbusConnect cbConnect = nullptr;
      |  ^~~~~~~~~~~~~~~
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h:52:2: error: 'cbModbusConnect' does not name a type
   52 |  cbModbusConnect cbDisconnect = nullptr;
      |  ^~~~~~~~~~~~~~~
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h:113:17: error: 'cbModbusConnect' has not been declared
  113 |  void onConnect(cbModbusConnect cb = nullptr);
      |                 ^~~~~~~~~~~~~~~
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h:114:20: error: 'cbModbusConnect' has not been declared
  114 |  void onDisconnect(cbModbusConnect cb = nullptr);
      |                    ^~~~~~~~~~~~~~~
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h: In member function 'void ModbusTCPTemplate<SERVER, CLIENT>::task()':
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h:227:8: error: 'cbConnect' was not declared in this scope; did you mean 'connect'?
  227 |    if (cbConnect == nullptr || cbConnect(currentClient->remoteIP())) {
      |        ^~~~~~~~~
      |        connect
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h: At global scope:
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h:424:6: error: variable or field 'onConnect' declared void
  424 | void ModbusTCPTemplate<SERVER, CLIENT>::onConnect(cbModbusConnect cb) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h:424:51: error: 'cbModbusConnect' was not declared in this scope
  424 | void ModbusTCPTemplate<SERVER, CLIENT>::onConnect(cbModbusConnect cb) {
      |                                                   ^~~~~~~~~~~~~~~
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h:429:6: error: variable or field 'onDisconnect' declared void
  429 | void ModbusTCPTemplate<SERVER, CLIENT>::onDisconnect(cbModbusConnect cb) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h:429:54: error: 'cbModbusConnect' was not declared in this scope
  429 | void ModbusTCPTemplate<SERVER, CLIENT>::onDisconnect(cbModbusConnect cb) {
      |                                                      ^~~~~~~~~~~~~~~
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h: In member function 'void ModbusTCPTemplate<SERVER, CLIENT>::cleanupConnections()':
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusTCPTemplate.h:441:8: error: 'cbDisconnect' was not declared in this scope; did you mean 'onDisconnect'?
  441 |    if (cbDisconnect && cbEnabled)
      |        ^~~~~~~~~~~~
      |        onDisconnect
In file included from C:\Users\kimwoo0.KOMECO\Documents\Arduino\sketch_jul10a\sketch_jul10a.ino:10:
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusEthernet.h: At global scope:
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusEthernet.h:15:53: error: expected class-name before '{' token
   15 | class EthernetServerWrapper : public EthernetServer {
      |                                                     ^
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusEthernet.h: In constructor 'EthernetServerWrapper::EthernetServerWrapper(uint16_t)':
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusEthernet.h:17:44: error: class 'EthernetServerWrapper' does not have any field named 'EthernetServer'
   17 |     EthernetServerWrapper(uint16_t port) : EthernetServer(port) {
      |                                            ^~~~~~~~~~~~~~
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusEthernet.h: At global scope:
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusEthernet.h:25:82: error: 'EthernetClient' was not declared in this scope; did you mean 'EthernetClass'?
   25 | class ModbusEthernet : public ModbusAPI<ModbusTCPTemplate<EthernetServerWrapper, EthernetClient>> {
      |                                                                                  ^~~~~~~~~~~~~~
[... cut ...]
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusEthernet.h:25:82: error: template argument 2 is invalid
c:\Users\kimwoo0.KOMECO\Documents\Arduino\libraries\modbus-esp8266\src/ModbusEthernet.h:25:96: error: template argument 1 is invalid
   25 | class ModbusEthernet : public ModbusAPI<ModbusTCPTemplate<EthernetServerWrapper, EthernetClient>> {
      |                                                                                                ^~

exit status 1

Compilation error: exit status 1

hacesoft, 2024-12-03:
The error you're encountering is due to the compiler being unable to find the definition for the types Callback or cbModbusConnect. This is typically caused by incorrect order of including header files or missing definitions.

Here is a step-by-step approach to resolve this issue:

1. Check the Order of Including Header Files
Ensure that all necessary header files are included in the correct order. Some definitions may depend on other header files that need to be included earlier.

2. Check Type Definitions
Ensure that the types Callback and cbModbusConnect are correctly defin