emelianov/modbus-esp8266 #359: modbusRTU with STM32L4

Rumen33, 2024-07-22:
I am trying to use the modbus RTU library with STM32L431RTC6 processor and it looks like the pulse on the direction pin of the processor is not long enough to achieve successful data transmition. I am using v. 4.1.0 of the library, but I also tried 4.1.1, without solving the problem. Any support will be appreciated. The code was also tested with ESP32 and works great there.

Rumen33, 2024-07-22:
I succeed to solve the problem by adding the following into ModbusSettings.h:

#if defined(ARDUINO_GENERIC_L431RCTX)   
#define MODBUSRTU_FLUSH_DELAY 20                    
#endif

Is this correct solution?

emelianov, 2024-10-27:
Requirements of manipulating with *MODBUSRTU_FLUSH_DELAY* indicates that Serial.flush() returns prior to serial buffer to send to line. I'm afraid no quick way for fix its properly without deep dive to STM hardware|SDK.