emelianov/modbus-esp8266 #364: Reading Holding Register with length of 2 and unit id 3

MeisterQ, 2024-10-01:
```
/*
  Modbus-Arduino Example - Master Modbus IP Client (ESP8266/ESP32)
  Read Holding Register from Server device

  (c)2018 Alexander Emelianov (a.m.emelianov@gmail.com)
  https://github.com/emelianov/modbus-esp8266
*/

#ifdef ESP8266
 #include <ESP8266WiFi.h>
#else
 #include <WiFi.h>
#endif
#include <ModbusIP_ESP8266.h>

const int REG = 30775;               // Modbus Hreg Offset
IPAddress remote(192, 168, 188, 152);  // Address of Modbus Slave device
const int LOOP_COUNT = 10;

ModbusIP mb;  //ModbusIP object

void setup() {
  Serial.begin(115200);
 
  WiFi.begin("---", "------");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  mb.client();
}

uint16_t res = 0;
uint8_t show = LOOP_COUNT;

void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
 // Serial.print("Conencted to Modbus server ");
    mb.readHreg(remote, 30775, &res, 3);  // Initiate Read Hregs from Modbus Server

  } else {
    mb.connect(remote);           // Try to connect if no connection
    Serial.println("Not connected");
  }
  mb.task();                      // Common local Modbus task
  delay(100);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    Serial.println(res);
    show = LOOP_COUNT;
  }
}
```

Im trying to connect to an Inverter and read register 30775 with the length of 2. But it gives me back 0. If i connect with qmodmaster it shows a value.

Device got Unit Id 3. I dont exactly get how you define it in mb.readHred. Ive i look at the describtion i need to fill more informations. Like register length, a NULL and so on?

[image]

[image]

emelianov, 2024-10-27:
I suspect right code snippet should be like that. Note that real register address offset is 775 (try 774, back to 30775 or 30774 if no success).

```cpp
uint16_t res[2] = {0,0};
uint8_t show = LOOP_COUNT;

void loop() {
  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
 // Serial.print("Conencted to Modbus server ");
    mb.readHreg(remote, 775, &res, 2, null, 3);  // Initiate Read Hregs from Modbus Server
  } else {
    mb.connect(remote);           // Try to connect if no connection
    Serial.println("Not connected");
  }
  mb.task();                      // Common local Modbus task
  delay(100);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    Serial.println(res);
    show = LOOP_COUNT;
  }
}
```

pflegefall, 2024-11-13:
I have the same issue, trying to read Hreg 30775 from SMA Inverter.
I used the original example as well as the above posted snippet emelianov, and tried 775, 774, 30775 and 30774.

I can read the Hreg with another library but want to switch over to this one.
Any further Ideas?

emelianov, 2024-11-24:
Attempt to utilize code based on the provided example: [this example](https://github.com/emelianov/modbus-esp8266/blob/master/examples/Callback/Request/Request.ino). This example will display the error code associated with the operation, which may provide additional insights into the underlying issue.

rberkelm, 2024-12-30:
As those before me, I need to read 2 registers on an inverter with slave address 247. I have success with ModbusPoll, but not yet with this library. My code snippet is:`
void loop() {
  uint16_t res[2] = {0,0};
  uint8_t show = LOOP_COUNT;

  if (mb.isConnected(remote)) {   // Check if connection to Modbus Slave is established
    mb.readHreg(remote, REG, &res ,2,null,247);  // Initiate Read Coil from Modbus Slave
  } else {
    Serial.println("Modbus slave NOT connected");  
    mb.connect(remote, port);           // Try to connect if no connection
  }
  mb.task();                      // Common local Modbus task
  delay(1000);                     // Pulling interval
  if (!show--) {                   // Display Slave register value one time per second (with default settings)
    Serial.println(res);
    show = LOOP_COUNT;
  }
}`
I get a 2errors: "Modbus_test_Sigenegy:70:20: error: invalid conversion from 'uint16_t*' {aka 'short unsigned int*'} to 'long long unsigned int' [-fpermissive]
   70 |     Serial.println(res);
      |                    ^~~
      |                    |
      |                    uint16_t* {aka short unsigned int*}
exit status 1
'null' was not declared in this scope"
I know what I'm doing wrong is probably pretty basic, but I don't know what the fix is. Any chance of showing a working example?

rberkelm, 2025-01-12:
I'm still struggling to read two consecutive registers. I keep getting a "no matching function for call to 'ModbusIP" error during compilation, due to data types. Basically the library won't accept "uint16_t res[2] = {0,0};"
I know it is a big ask, but since others are stuck on this as well, would you be able to publish a working example for reading 2 consecutive registers?

rberkelm, 2025-01-12:
Never mind, I found the answer [here](https://github.com/emelianov/modbus-esp8266/issues/46).
The trick is to read the Hreg with: 
`mb.readHreg(remote, REG, (uint16_t*)res,2,cb,247);`
instead of
`mb.readHreg(remote, REG, res,2,cb,247);`