emelianov/modbus-esp8266 #369: stuck with Master and Slace===

pacraf, 2024-10-29:


pacraf, 2024-10-29:
As expected, it was my mistake, I did not setup slave address
adding this
mb2.slave(SLAVE_ID);

pacraf, 2024-10-29:
If I could ask for explanation of this part of code:
mb1.pullHreg(PULL_ID, FROM_REG, TO_REG,2);
My understanding is that it is pulling registers from slave device starting from register "FROM_REG" and writes it under register "TO_REG" , and register count is 2 in this example. That makes in this basic example the case that "ESP32 - Slave RTU" properly responds the same values that were just read by "ESP32 - Master RTU" from Energy meter.  
and my question is - what is limit of registers count? 
should it not be defined in advance if I would like to take 90 registers from this meter? 
or  it is just fine to 
mb1.pullHreg(PULL_ID, FROM_REG, TO_REG,90);
and not worry

emelianov, 2024-11-04:
> mb1.pullHreg(PULL_ID, FROM_REG, TO_REG,90);

Works exactly as expected reading 90 sequential registers. `mb1.pullHreg(PULL_ID, FROM_REG, TO_REG)` reads just one register.

pacraf, 2024-11-05:
Thank you for confirming. 
basically program works (as bridge RTU <>RTU) but I think that it is not as should be. 

please advice on cleanup and understand what is really going on in loop
What is my objective:
1. that the loop will periodically pull from slave 3 sets of registers. (defined as FROMxxxx, with different legth) 
I see that it is happening, when I monitor mb1 (serial1) I see frames are flying.

And please here advice how a lean loop should do this task? like in my code below ?
3 times such sequence (for all 3 ranges)? adding delay is ok? 
```cpp
if(!mb1.slave())
    mb1.pullHreg(1, FROM_REG, TO_REG,90);
  mb1.task();
  mb2.task();
  yield();
  delay(10); 
```
2. Another thing I do not understand is that in this code below, I try during this bridge work to read two registers and convert to float. 
this part of code below should also check in slave is not occupied by communication?
```cpp
if(!mb1.slave()){
  uint16_t A0 = mb1.Hreg(0x2006);
  uint16_t A1 = mb1.Hreg(0x2007);
```
Converting works, but behavior is that in fact, after reset, lets assume that ESP slave RTU was not asked for anything. But ESP Master RTU constantly asks for all 3 registers sets. But my reading of this register is showing zeros. But if I start talking to ESP slave RTU for values in these registers, it responds correctly (bridge function works) and THEN from, this moment where this communication initiated, I also see values in registers, and my float appears is debug output. 

so definetely I am doing it not correctly. 
The question is for this example bridge, what is good way to access registers that are  periodically questioned by ESP Master RTU . They are readed (I see communication) but how to access them properly... ?

3. I also have some doubts if it is OK for this specific bridge idea to copy and paste the same register addresses... you see below, that for example FROM_BASE is the same as TO_BASE. maybe this is not correct at all...  My bridge should be a "MIRROR Bridge" , the same addresses are for both master and slave part of ESP RTU. 

```cpp

#include <ModbusIP_ESP8266.h>
#include <ModbusRTU.h>

#define TO_REG 0x1000
#define TO_UABC_REG 0x2006
#define TO_BASE 0x0000
#define SLAVE_ID 1
#define PULL_ID 1
#define FROM_BASE 0x0000
#define FROM_REG 0x1000
#define FROM_UABC_REG 0x2006

ModbusRTU mb1;
ModbusRTU mb2;

void setup() {

 
//  

  Serial2.begin(9600, SERIAL_8N1, 33, 32); //Init Serial2  
  Serial1.begin(9600, SERIAL_8N1, 19, 18); // Init Serial1 
  Serial.begin(115200, SERIAL_8N1); // default pins for ESP32 debug

  mb2.begin(&Serial2, 27); // slave = emu chint
  mb1.begin(&Serial1, 17);  // master for chint on Serial1

 // Serial.println("setup");
  mb1.master();   //master for  chint
  mb2.server(); //server = slave = emulator of chint
  mb2.slave(SLAVE_ID);
  mb2.addHreg(TO_REG);
  mb2.addHreg(TO_UABC_REG);
  mb2.addHreg(TO_BASE);
}
float value32;
float EnergyEX;
float UA;
float UB;
float UC;
float Energy4;
float Energy5;
float Energy6;
//int32 value32;

float f_2uint_float(unsigned int uint1, unsigned int uint2) {    // reconstruct the float from 2 unsigned integers

  union f_2uint {
    float f;
    uint16_t i[2];
  };
  union f_2uint f_number;
  f_number.i[0] = uint1;
  f_number.i[1] = uint2;
  return f_number.f;
}
void loop() {
  if(!mb1.slave())
    mb1.pullHreg(1, FROM_REG, TO_REG,90);
  mb1.task();
  mb2.task();
  yield();
  delay(10); 
  Serial.println("loop");
  if(!mb1.slave())
    mb1.pullHreg(1, FROM_UABC_REG, TO_UABC_REG,40);
  mb1.task();
  mb2.task();
  yield();
  delay(10);
  if(!mb1.slave()){
  uint16_t A0 = mb1.Hreg(0x2006);
  uint16_t A1 = mb1.Hreg(0x2007);
  UA = f_2uint_float(A1,A0);
  Serial.print(UA);
  }
  
  delay(10);
  if(!mb1.slave())
    mb1.pullHreg(1, FROM_BASE, TO_BASE,10);

  mb1.task();
  mb2.task();
  yield();
  delay(10); 
}
```

closed 2024-11-08 by pacraf