emelianov/modbus-esp8266 #370: Timeout error (0xE4) when requesting more than 28 registers.

flintstones3000, 2024-11-10:
First of all thanks for developing this library.  It has been working great on my ESP8266.  I've come across an issue when requesting a read of more than 28 registers returns a timeout error message (0xE4).  I've tried increasing **#define MODBUSRTU_TIMEOUT 10000** in ModbusSetting.h but still no luck.  Am I doing something wrong?

In my code below I'm requesting 30 registers:
`/*
*/

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <StreamString.h>
#include <SoftwareSerial.h>
#include <ModbusRTU.h>

#ifndef STASSID
#define STASSID "SSID"
#define STAPSK "PSK"
#endif

#if defined(ESP8266)
#define D8 (15)
#define D5 (14)
#define D7 (13)
#define D6 (12)
#endif

const byte Start_Reg = 1;
const char *ssid = STASSID;
const char *password = STAPSK;
const int UnitID = 2;
const int led = 13;

long int  Val_SerialNumber = 0;

uint16_t holdingRegisters[62];
unsigned long combined = 0; // clear it out
float power = 0;
union equiv {
  float x;
  unsigned long l;
} equiv ;

ESP8266WebServer server(80);
SoftwareSerial S(D5, D6);
ModbusRTU mb;

bool cb(Modbus::ResultCode event, uint16_t transactionId, void* data) {
  
  equiv.l = ((long)holdingRegisters[0] << 16) | (holdingRegisters[1]);
  Val_SerialNumber = equiv.l;
  Serial.println(Val_SerialNumber);
  
  if (event != Modbus::EX_SUCCESS) {
    Serial.print("Request result: 0x");
    Serial.print(event, HEX); 
  }
  return true;
}

void handleRoot() {
  digitalWrite(led, 1);
  
  mb.readHreg(UnitID, Start_Reg, &holdingRegisters[0], 30, cb);
  while(mb.slave()){
    mb.task();
    delay(10);
  }

  digitalWrite(led, 0);
}

void handleNotFound() {
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";

  for (uint8_t i = 0; i < server.args(); i++) { message += " " + server.argName(i) + ": " + server.arg(i) + "\n"; }

  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void) {
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");
  
  S.begin(9600, SWSERIAL_8N1);
  mb.begin(&S, D7);    //RE Pin
  mb.master();

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) { Serial.println("MDNS responder started"); }

  server.on("/", handleRoot);
  server.onNotFound(handleNotFound);
  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
  MDNS.update();
}`

emelianov, 2024-11-20:
Try to play with SoftwareSerial library versions.
Also you can try to change power source for the board,

RosiersRobin, 2025-01-10:
I get this error always, no mater what. Reading a single input register... I can read out the device with modbus poll on a windows device using brainboxes serial connector. I've never seen this error before and I'm using a really short cable (like max 1m)