emelianov/modbus-esp8266 #372: Unit_ID definition

hacesoft, 2024-11-23:
I have this code:

`#include <ESP8266WiFi.h>
#include <ModbusIP_ESP8266.h>

// Modbus Registers Offsets
const int TEST_HREG = 100;
#define LED_PIN D4 // Definice pinu pro LED diodu
#define Hi 1
#define Lo 0
#define cPeriodaLED 10 // Perioda pro blikání LED v milisekundách (0.5 sekundy)
#define Port_ModbusIP 503
#define Unit_Id 100

#define SSID "xxx"
#define PASS "xxx"

bool bStatusResetDefault = Lo;
bool bDefaul_PIN_Reset = Lo;
bool bSwitch_LED_Status = true;
bool ledBlinking = true;
int iCounter = 0;
unsigned long previousMillis = 0; // Předchozí čas pro časování
const long interval = 1000; // Interval pro generování nového čísla (1 sekunda)

// ModbusIP object
ModbusIP modbusTCPServer;

//**************************************************************************************
void Blikani() {
  // Blikání LEDky, to jen abych věděl, že HW žije...
  if (bSwitch_LED_Status == true) {  // když je tlačítko stisknuto
    iCounter++;
    if (iCounter == cPeriodaLED / 2) {
      digitalWrite(LED_PIN, Hi);
    } else if (iCounter == cPeriodaLED) {
      digitalWrite(LED_PIN, Lo);
      iCounter = 0;
    }
  } else {  // Vypnutí LEDky
    iCounter = 0;
    digitalWrite(LED_PIN, Lo);
  }
}

void setup() {
  Serial.begin(115200);

  // Nastavení pinu pro LED diodu jako výstup
  pinMode(LED_PIN, OUTPUT);

  WiFi.begin(SSID, PASS);

  // Nastavení pevné IP adresy, masky sítě a brány
  IPAddress ip(192, 168, 11, 4);
  IPAddress subnet(255, 255, 255, 0);
  IPAddress gateway(192, 168, 11, 1);

  // Připojení k Wi-Fi s pevným nastavením IP
  WiFi.config(ip, gateway, subnet);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  // Nastavení ModbusIP jako server
  modbusTCPServer.server(Port_ModbusIP);
  modbusTCPServer.addHreg(TEST_HREG, 0); // Nastavení registru
}

void loop() {
  // Volání funkce Blikani()
  Blikani();
  // Zpracování Modbus úkolů
  modbusTCPServer.task();
  // Použití yield() pro zabránění resetu watchdog timeru
  yield();

  // Generování náhodného čísla a zápis do holding registru v sekundových intervalech
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;
    uint16_t randomValue = random(0, 0xFFFF); // Generování náhodného čísla od 0 do 0xFFFF
    modbusTCPServer.Hreg(TEST_HREG, randomValue); // Zápis náhodného čísla do holding registru
    Serial.print("Writing random value to holding register: ");
    Serial.println(randomValue, HEX);
  }

  delay(100);
}
`  I made a modbus.TCP server in NodeMCU. How do I set the Unit_ID, I couldn't. It's still set to 1 by default. I'm reading data in Node-RED and I want to define internal registers, currently I only have one but I don't know how to change the Unit_ID. 

[image]

emelianov, 2024-11-24:
The implementation of Modbus TCP Server disregards the UnitID and response values in requests, regardless of the UnitID value.
There are alternative methods to handle the UnitID value, but I would appreciate it if you could clarify your specific objective.

hacesoft, 2024-11-24:
> The implementation of Modbus TCP Server disregards the UnitID and response values in requests, regardless of the UnitID value. There are alternative methods to handle the UnitID value, but I would appreciate it if you could clarify your specific objective.

I want this to simplify the code in node-red, when I communicate with a victron inverter, it has all registers on UnitID 100, I would like to use one template for connection and only change the IP address. Now I have to change the IP address and UnitID. It is more for pragmatic reasons. I am trying to write the code in Node-Red so that after uploading Flow the user does not have to change anything in the Flow settings, does not have to define any variables when starting the environment - some implementations do not allow this at all, for example Cerbo for controlling the inverter. If Node-Red runs in a container, you can define the variables however you want. I want to use only one template for communication for ModBus, now the one I use for communication with Victron inverters has the IP set to some LOOP loop and I change it when starting Flow and the user sets his IP address in the UI. So I could use the same template for NodeMCU where I will also read data from Node-Red... It's nothing dramatic, but I would also welcome a change to the UnitID

hacesoft, 2024-12-25:
I've already solved the problem in a different way, so there's probably no need to worry about it.

closed 2024-12-25 by hacesoft